libusb/hidapi #620: hidapi_darwin.h is not installed

ryandesign, 2023-09-03:
When using the autotools build system on macOS, the hidapi_darwin.h header is not installed. The only installed header is hidapi.h. I am aware that the autotools build system is deprecated.

Youw, 2023-09-04:
> I am aware that the autotools build system is deprecated.

Than you should understand why I'm closing this one.

closed 2023-09-04 by Youw

mcuee, 2023-09-04:
@Youw 

Maybe we want to remove the auto-tools stuff in the next version so that people will not get confused.

Youw, 2023-09-04:
> Maybe we want to remove the auto-tools stuff in the next version so that people will not get confused.

The justification for #622 has some point. I'll have that discussion first.
And again - we've had that discussion in the past and agreed to obsolete Autotools completely when API v1 is developed. And that is a huge piece of work.
Removing Autotools completely will be very strong motivation to move to CMake, but I can imagine that it not always easy/possible for some users.

ryandesign, 2023-09-04:
> Than you should understand why I'm closing this one.

That's up to you. I'm a developer with MacPorts. When we find problems in the software that we have in our collection, we try to report those problems to the upstream developers so that they can take whatever action they deem best.