libusb/hidapi #628: hid_enumerate prints usage of all devices

Mola19, 2023-09-26:
This might be user error by me, or maybe it is already solved, but i couldn't find a similar issue.

My code is

``` c++
#include <iostream>

#include "hidapi.h"

int main()
{
    hid_enumerate(0, 0);
    std::cout << "---END---" << std::endl;
}
```

Output:

```
  ---END---
6
6
1
6
1
80
1
1
1
1
92
6
1
6
2
2
1
6
E
1
1
1
6
A1
80
5
80
1
1
1
A1
1
1
  ```

The numbers seem to be the usage for each hid device. 

The weird thing is that they are printed after the code ends.

Whether `hid_init()`, `hid_exit()` or `hid_free_enumeration` get called doesn't influence this issue.

OS: Windows 10
Compiler: MSVC C++ 17
HIDAPI version: I reproduced this on 0.14.0, 0.11.0 and 0.9.0

Youw, 2023-09-26:
very interesting
winapi backend doesn't even have the `printf` anywhere
is the behavior changes if you you `printf` instead of `cout`?

Mola19, 2023-09-26:
No, it doesn't seem to change anything

Mola19, 2025-10-25:
So I have looked at this again and it seems that I had an hidapi version in the path, that overwrote the version I tried to link. I don't remember doing it, but it seems likely that I built it myself with debug prints.
It works now with 0.16.0, but I cannot test older versions, since cmake claims it dropped support.
Anyway I think this issue was user error and if someone observes this behavior again they can reopen it.

closed 2025-10-25 by Mola19