libusb/hidapi #632: Fix MinGW CI build

Youw, 2023-10-23:
Recently MinGW caught an issue (several issues) in the test tool:
```sh
D:\a\hidapi\hidapi\hidapisrc\windows\test\hid_report_reconstructor_test.c:105:34: error: too many arguments for format [-Werror=format-extra-args]
  105 |                 if (sscanf(line, "pp_data->MagicKey                             = 0x%02hhX%02hhX%02hhX%02hhX%02hhX%02hhX%02hhX%02hhX\n", &pp_data->MagicKey[0], &pp_data->MagicKey[1], &pp_data->MagicKey[2], &pp_data->MagicKey[3], &pp_data->MagicKey[4], &pp_data->MagicKey[5], &pp_data->MagicKey[6], &pp_data->MagicKey[7])) continue;
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

Need to handle it.

JoergAtGithub, 2023-10-23:
If I look on the code, it looks like a false-positive warning to me. The format string has 8 arguments and the function call too.

Maybe some settings like __USE_MINGW_ANSI_STDIO or -std=c99 might help with MinGW: https://stackoverflow.com/questions/4101335/does-gccwindows-mingw-defines-scnd8-scnu8-in-inttypes-h/4102215#4102215

Youw, 2023-10-23:
might be related to https://github.com/libusb/libusb/pull/1327#issuecomment-1766992570

mcuee, 2023-11-16:
Hmm, I can not reproduce the issue with a fully up-to-date MSYS2 mingw64 build.

```
MINGW64 /c/work/hid/hidapi

$ cat build_verbose.sh
cmake -B build_mingw -D CMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DHIDAPI_WITH_TESTS=ON -DHIDAPI_BUILD_PP_DATA_DUMP=ON -DCMAKE_C_FLAGS="-Wall -Wextra -pedantic -Werror"
cmake --build build_mingw

$ . ./build_verbose.sh
-- Building for: Ninja
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/msys64/mingw64/bin/cc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- hidapi: v0.14.0
-- Configuring done (0.8s)
-- Generating done (0.0s)
-- Build files have been written to: C:/work/hid/hidapi/build_mingw
Change Dir: 'C:/work/hid/hidapi/build_mingw'

Run Build Command(s): C:/msys64/mingw64/bin/ninja.exe -v
[1/8] C:\msys64\mingw64\bin\windres.exe -O coff -Dhidapi_winapi_EXPORTS -I C:/work/hid/hidapi/hidapi -I C:/work/hid/hidapi/windows  C:/work/hid/hidapi/windows/hidapi.rc src/windows/CMakeFiles/hidapi_winapi.dir/hidapi.rc.obj
[2/8] C:\msys64\mingw64\bin\cc.exe  -IC:/work/hid/hidapi/hidapi -IC:/work/hid/hidapi/windows -Wall -Wextra -pedantic -Werror -O2 -g -DNDEBUG -std=gnu11 -MD -MT src/windows/test/CMakeFiles/hid_report_reconstructor_test.dir/hid_report_reconstructor_test.c.obj -MF src\windows\test\CMakeFiles\hid_report_reconstructor_test.dir\hid_report_reconstructor_test.c.obj.d -o src/windows/test/CMakeFiles/hid_report_reconstructor_test.dir/hid_report_reconstructor_test.c.obj -c C:/work/hid/hidapi/windows/test/hid_report_reconstructor_test.c
[3/8] C:\msys64\mingw64\bin\cc.exe -Dhidapi_winapi_EXPORTS -IC:/work/hid/hidapi/hidapi -IC:/work/hid/hidapi/windows -Wall -Wextra -pedantic -Werror -O2 -g -DNDEBUG -MD -MT src/windows/CMakeFiles/hidapi_winapi.dir/hid.c.obj -MF src\windows\CMakeFiles\hidapi_winapi.dir\hid.c.obj.d -o src/windows/CMakeFiles/hidapi_winapi.dir/hid.c.obj -c C:/work/hid/hidapi/windows/hid.c
[4/8] C:\msys64\mingw64\bin\cc.exe -Dhidapi_winapi_EXPORTS -IC:/work/hid/hidapi/hidapi -IC:/work/hid/hidapi/windows -Wall -Wextra -pedantic -Werror -O2 -g -DNDEBUG -MD -MT src/windows/CMakeFiles/hidapi_winapi.dir/hidapi_descriptor_reconstruct.c.obj -MF src\windows\CMakeFiles\hidapi_winapi.dir\hidapi_descriptor_reconstruct.c.obj.d -o src/windows/CMakeFiles/hidapi_winapi.dir/hidapi_descriptor_reconstruct.c.obj -c C:/work/hid/hidapi/windows/hidapi_descriptor_reconstruct.c
[5/8] cmd.exe /C "cd . && C:\msys64\mingw64\bin\cc.exe -Wall -Wextra -pedantic -Werror -O2 -g -DNDEBUG   -shared -o src\windows\libhidapi.dll -Wl,--out-implib,src\windows\libhidapi.dll.a -Wl,--major-image-version,0,--minor-image-version,14 src/windows/CMakeFiles/hidapi_winapi.dir/hid.c.obj src/windows/CMakeFiles/hidapi_winapi.dir/hidapi_descriptor_reconstruct.c.obj src/windows/CMakeFiles/hidapi_winapi.dir/hidapi.rc.obj  -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 && cd ."
[6/8] cmd.exe /C "cd . && C:\msys64\mingw64\bin\cc.exe -Wall -Wextra -pedantic -Werror -O2 -g -DNDEBUG  src/windows/test/CMakeFiles/hid_report_reconstructor_test.dir/hid_report_reconstructor_test.c.obj -o src\windows\test\hid_report_reconstructor_test.exe -Wl,--out-implib,src\windows\test\libhid_report_reconstructor_test.dll.a -Wl,--major-image-version,0,--minor-image-version,0  src/windows/libhidapi.dll.a  -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 && cd ."
[7/8] C:\msys64\mingw64\bin\cc.exe  -IC:/work/hid/hidapi/hidapi -IC:/work/hid/hidapi/windows -Wall -Wextra -pedantic -Werror -O2 -g -DNDEBUG -std=gnu11 -MD -MT src/windows/pp_data_dump/CMakeFiles/pp_data_dump.dir/pp_data_dump.c.obj -MF src\windows\pp_data_dump\CMakeFiles\pp_data_dump.dir\pp_data_dump.c.obj.d -o src/windows/pp_data_dump/CMakeFiles/pp_data_dump.dir/pp_data_dump.c.obj -c C:/work/hid/hidapi/windows/pp_data_dump/pp_data_dump.c
[8/8] cmd.exe /C "cd . && C:\msys64\mingw64\bin\cc.exe -Wall -Wextra -pedantic -Werror -O2 -g -DNDEBUG  src/windows/pp_data_dump/CMakeFiles/pp_data_dump.dir/pp_data_dump.c.obj -o src\windows\pp_data_dump\pp_data_dump.exe -Wl,--out-implib,src\windows\pp_data_dump\libpp_data_dump.dll.a -Wl,--major-image-version,0,--minor-image-version,0  src/windows/libhidapi.dll.a  -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 && cd ."
```

PR #641 (2023-11-17): Fix the windows build by fixing printf related issues
This merge-request removes the need for 
```C
    #pragma GCC diagnostic ignored "-Wformat"
    #pragma GCC diagnostic ignored "-Wformat-extra-args"
```
in mingw. It also turns on `-Wformat-signedness` for GNU-compatible compilers.

This fixes #632
closed 2023-11-17 by Youw