libusb/hidapi #650: Hang forever in hid_read_timeout() when calling hid_close().

dontech, 2023-11-28:
Hello,

There have been previous reports on this:

https://github.com/signal11/hidapi/issues/88
https://github.com/libusb/hidapi/issues/128

These are assumed fix, but they are not.

I have simple cases that hang 100% of the time:

1) Create reader in thread using hid_read_timeout()
2) Call hid_close() in main thread
3) GetOverlappedResult() blocks forever.

dontech, 2023-11-28:
I discovered that CancelIoEx() has secret properties that can shutdown overlapped operations that CancelIo() cannot for some odd reason.

According to docs, CancelIo() should do the same, but I fear that for reason the system cannot find the overlapped operation from the handle when called. By calling CancelIoEx() directly with the overlapped reference, the problem is resolved.

Pull request sent.

With this fix the problem becomes un-reproducible in my case. 

Thanks,

/pedro

mcuee, 2023-11-28:
@dontech 

Thanks. Is your issue the same as #133? Your fix seems to be the same as mentioned in #133 as well -- to use `CancelIoEx`.

BTW, I do not see any issues to drop Windows XP support. But @Youw will need to agree.

* #133 
* https://github.com/libusb/hidapi/issues/237

mcuee, 2023-11-28:
Related discussion:
* #649

Youw, 2023-11-29:
Closing as duplicate of #133

closed 2023-11-29 by Youw

mcuee, 2023-11-29:
> > BTW, I do not see any issues to drop Windows XP support. But @Youw will need to agree.
> 
> Again, it is not about WindowsXP support, but rather to catch early impropper use of the API.

What I mean is that if the fix to use CancelEx is good then there is no problem is drop XP support. But since that is not a vliad fix, that is another story.