libusb/hidapi #655: ImportError of usblib libraries

rolsman1974, 2023-12-30:
Hi,
On my Mac it is very difficult to get the hid / hidapi to work, either you get errors the usblib1 backends does not support enumeration or you get ImportErrors with regards to the usblib's

I also can find a good manual how to install these packages and to set things up correctly.
Is there a specific reason why running 'just' pip install hid , get your system ready to go?

My searches on internet didn't resolve my issue getting things to work, it did tell me lots of people facing this issue.

My request please
- just make using pip install only to set things up correctly
  or
- provide a good manual to set things up

Kind regards

JoergAtGithub, 2023-12-30:
In this project we develop the C library hidapi - it seems you have problems with a Python wrapper, which comes from a third-party.

JoergAtGithub, 2023-12-30:
It might be this Python wrapper, that you use: https://github.com/trezor/cython-hidapi

mcuee, 2023-12-30:
Please raise the issue to respective project. hidapi project does not handle the Python wrapper.

closed 2023-12-30 by mcuee