libusb/hidapi #661: Meson build system cannot be used for cross builds

amyspark, 2024-02-29:
Hi,

This is to report that CMake-based subprojects, like the one hidapi's Meson build system uses, do not work for cross builds: https://github.com/mesonbuild/meson/issues/12761

We found this issue when attempting to cross compile https://github.com/blackmagic-debug/blackmagic [from arm64 AppleClang to arm64-none-eabi-gcc](https://files.xobs.io/file/XAszURPYyXBwEUMs/AZzdyuO9IYNO1deQ/meson-log.txt) (see towards the end).

Youw, 2024-02-29:
Is this something that  may be fixed by modifying https://github.com/libusb/hidapi/blob/master/meson.build + https://github.com/libusb/hidapi/blob/master/subprojects/hidapi_build_cmake/CMakeLists.txt or is this a general limitation of https://mesonbuild.com/CMake-module.html ?

amyspark, 2024-02-29:
Neither, Meson currently just doesn't pass the correct variables in the toolchain file (it passes the build machine's and not the host's). The only way we know to fix it is turn the meson.build into a complete port.

Youw, 2024-02-29:
So we're back to the question "who is going to support it?".
Multiple supported build systems is not something any sain maintainer would sign up just for convenience.
The community is free to have its own implementation of Mason build system for HIDAPI.
Similar way to how https://github.com/trezor/cython-hidapi exists.

Youw, 2024-02-29:
Closing as a limitation of Meson's [CMake-module.html](https://mesonbuild.com/CMake-module.html).

closed 2024-02-29 by Youw