libusb/hidapi #666: Potential solution to MacOS hid_init/hid_exit issue

k1-801, 2024-03-11:
One of the known thread-safety-related limitations on MacOS is that hid_init and hid_exit have to be called on the same thread. The only thing hid_exit really does there is destroy a HidManager. We could just leave that on the end user's shoulders, or we could resolve it internally.

Theoretically, this issue could be easily mitigated by giving the manager it's own thread to live in, which apparently would always be suspended by the sleeping RunLoop waiting for events. The manager would be created AND destroyed within the thread's function, guaranteed to be in the same thread. hid_init would have to start the thread and then wait at a barrier for the manager to be created. Then hid_exit would essentially be built the same way as hid_close, sending an event to stop the thread. That's about it.

Youw, 2024-04-06:
All true.

I was thinking about having its own thread for init/deinit on macOS.
Just never had a chance to get it done.

tresf, 2024-06-07:
> We could just leave that on the end user's shoulders

Ran into this recently after some new threading was added to a downstream application.  We decided to wrap calls documented here https://github.com/libusb/hidapi/issues/45#issuecomment-1546803030 into a single thread, which seems to have fixed the issue.  Since we're actually downstream of downstream, so we submitted the patch to the library that wraps hidapi functionality.  I'm not sure if the patch will be accepted, but I wanted to offer gratitude for documenting this.

marciopamplona, 2024-07-19:
Hello, reverting this change seems to fix the issue. Any drawbacks on doing this ?

[image]

Youw, 2024-07-21:
If will break other scenarios, not similar to yours.

xobs, 2025-09-04:
Thanks. I appreciate that input.

Is there a way around it? The issue I see right now is that you need to know what thread first called into hidapi and then deallocate threadapi if that thread quits.

* Is there a problem with initializing hidapi multiple times? Or is it a singleton?
* Is there a way that you know of to fire off a runloop in another thread on init? It might be better to waste a thread rather than have mysterious crashes...
* What are the implications of calling `hid_exit()` while other threads are still using it?

I'm trying to come up with a way to track usage across multiple threads in the Rust library, and I'm not really coming up with anything foolproof other than documentation saying to always call into hidapi first from the main thread, which doesn't seem ideal...

Youw, 2025-09-04:
> Is there a problem with initializing hidapi multiple times? Or is it a singleton?

Global initialisation is a singletone.

> Is there a way that you know of to fire off a runloop in another thread on init? It might be better to waste a thread rather than have mysterious crashes...

https://github.com/libusb/hidapi/issues/666#issuecomment-2041026698

> What are the implications of calling hid_exit() while other threads are still using it?

This is undefined behavior. (Depends on a platform, but most likely it will crash at some point, maybe immediately)

---
See also: https://github.com/libusb/hidapi/wiki/Multi%E2%80%90threading-Notes

> Is there a way around it?

The "ideal workaround" (sounds funny, right?) is to create a dedicated thread, and call hid_init/hid_exit (maybe even all other "global" functions from that thread. As I mentioned in my other comment - probably best if that thread is created by hidapi itself, but I don't find that to be trivial being done in HIDAPI source right now in C.
Besides, we have a single global state right now, and a thread per-device. I'd like that to be changed for v1, anyway.