libusb/hidapi #669: Cannot compile hidapi 1.5.0 on macos

so-schen, 2024-03-15:
error: failed to run custom build command for `hidapi v1.5.0`

Caused by:
  process didn't exit successfully: `/Users/kaiqichen/Documents/share/repo/aptos-core/target/debug/build/hidapi-59d4b98bcd965422/build-script-build` (exit status: 1)
  --- stdout
  TARGET = Some("aarch64-apple-darwin")
  OPT_LEVEL = Some("0")
  HOST = Some("aarch64-apple-darwin")
  CC_aarch64-apple-darwin = None
  CC_aarch64_apple_darwin = None
  HOST_CC = None
  CC = Some("/opt/homebrew/opt/llvm@13/bin/clang")
  CFLAGS_aarch64-apple-darwin = None
  CFLAGS_aarch64_apple_darwin = None
  HOST_CFLAGS = None
  CFLAGS = None
  CRATE_CC_NO_DEFAULTS = None
  DEBUG = Some("true")
  running: "/opt/homebrew/opt/llvm@13/bin/clang" "-O0" "-ffunction-sections" "-fdata-sections" "-fPIC" "-g" "-fno-omit-frame-pointer" "--target=arm64-apple-darwin" "-I" "etc/hidapi/hidapi" "-Wall" "-Wextra" "-o" "/Users/kaiqichen/Documents/share/repo/aptos-core/target/debug/build/hidapi-1db1005f8c948514/out/etc/hidapi/mac/hid.o" "-c" "etc/hidapi/mac/hid.c"
  cargo:warning=In file included from etc/hidapi/mac/hid.c:25:
  cargo:warning=In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDManager.h:29:
  cargo:warning=In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDLib.h:34:
  cargo:warning=/Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDLibObsolete.h:197:5: error: type name requires a specifier or qualifier
  cargo:warning=    IUNKNOWN_C_GUTS;
  cargo:warning=    ^
  cargo:warning=/Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDLibObsolete.h:204:5: error: type name requires a specifier or qualifier
  cargo:warning=    IUNKNOWN_C_GUTS;
  cargo:warning=    ^
  cargo:warning=/Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDLibObsolete.h:211:5: error: type name requires a specifier or qualifier
  cargo:warning=    IUNKNOWN_C_GUTS;
  cargo:warning=    ^
  cargo:warning=/Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDLibObsolete.h:241:5: error: type name requires a specifier or qualifier
  cargo:warning=    IUNKNOWN_C_GUTS;
  cargo:warning=    ^
  cargo:warning=/Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDLibObsolete.h:272:5: error: type name requires a specifier or qualifier
  cargo:warning=    IUNKNOWN_C_GUTS;
  cargo:warning=    ^
  cargo:warning=5 errors generated.
  exit status: 1

  --- stderr

mcuee, 2024-03-16:
Please provide more details. It seems to me you are trying to build a rust binding of hidapi and not hidapi itself.

In that case, this is not a valid question and you need to post the issue to that project.

mcuee, 2024-03-16:
You may want to report the issue to https://github.com/aptos-labs/aptos-core.

mcuee, 2024-03-16:
Official hidapi repo is here. If you have issues building hidapi git or hidapi-0.14.0, then please post the build log again.
https://github.com/libusb/hidapi

closed 2024-03-16 by mcuee

iamqk, 2025-04-23:
I encounter similar error again when I try build [Cemu](https://github.com/cemu-project/Cemu) project.

Cemu cmake log:
---------------------------------------------------------------------------------------------------------
Installing 1/18 hidapi:x64-osx@0.14.0...
Building hidapi:x64-osx@0.14.0...
-- Using cached libusb-hidapi-hidapi-0.14.0.tar.gz
-- Cleaning sources at /Users/xxx/Documents/projects/Cemu/dependencies/vcpkg/buildtrees/hidapi/src/api-0.14.0-e052aaf522.clean. Use --editable to skip cleaning for the packages you specify.
-- Extracting source /Users/xxx/Documents/projects/Cemu/dependencies/vcpkg/downloads/libusb-hidapi-hidapi-0.14.0.tar.gz
-- Using source at /Users/xxx/Documents/projects/Cemu/dependencies/vcpkg/buildtrees/hidapi/src/api-0.14.0-e052aaf522.clean
-- Found external ninja('1.12.1').
-- Configuring x64-osx
CMake Warning at /Users/xxx/Documents/projects/Cemu/build/vcpkg_installed/x64-osx/share/vcpkg-cmake/vcpkg_cmake_configure.cmake:327 (message):
  The following variables are not used in CMakeLists.txt:

      HIDAPI_BUILD_PP_DATA_DUMP

  Please recheck them and remove the unnecessary options from the
  `vcpkg_cmake_configure` call.

  If these options should still be passed for whatever reason, please use the
  `MAYBE_UNUSED_VARIABLES` argument.
Call Stack (most recent call first):
  buildtrees/versioning_/versions/hidapi/ef7fae56bd9bec8a3b50d4ab3f87ba7537ea330b/portfile.cmake:14 (vcpkg_cmake_configure)
  scripts/ports.cmake:206 (include)

-- Building x64-osx-dbg
CMake Error at scripts/cmake/vcpkg_execute_build_process.cmake:134 (message):
    Command failed: /usr/local/bin/cmake --build . --config Debug --target install -- -v -j13
    Working Directory: /Users/xxx/Documents/projects/Cemu/dependencies/vcpkg/buildtrees/hidapi/x64-osx-dbg
    See logs for more information:
      /Users/xxx/Documents/projects/Cemu/dependencies/vcpkg/buildtrees/hidapi/install-x64-osx-dbg-out.log

Call Stack (most recent call first):
  /Users/xxx/Documents/projects/Cemu/build/vcpkg_installed/x64-osx/share/vcpkg-cmake/vcpkg_cmake_build.cmake:74 (vcpkg_execute_build_process)
  /Users/xxx/Documents/projects/Cemu/build/vcpkg_installed/x64-osx/share/vcpkg-cmake/vcpkg_cmake_install.cmake:16 (vcpkg_cmake_build)
  buildtrees/versioning_/versions/hidapi/ef7fae56bd9bec8a3b50d4ab3f87ba7537ea330b/portfile.cmake:19 (vcpkg_cmake_install)
  scripts/ports.cmake:206 (include)

error: building hidapi:x64-osx failed with: BUILD_FAILED
See https://learn.microsoft.com/vcpkg/troubleshoot/build-failures?WT.mc_id=vcpkg_inproduct_cli for more information.
Elapsed time to handle hidapi:x64-osx: 3.3 s
Please ensure you're using the latest port files with `git pull` and `vcpkg update`.
Then check for known issues at:
  https://github.com/microsoft/vcpkg/issues?q=is%3Aissue+is%3Aopen+in%3Atitle+hidapi
You can submit a new issue at:
  https://github.com/microsoft/vcpkg/issues/new?title=[hidapi]+Build+error+on+x64-osx&body=Copy%20issue%20body%20from%20%2FUsers%2Fkeqiao%2FDocuments%2Fprojects%2FCemu%2Fbuild%2Fvcpkg_installed%2Fvcpkg%2Fissue_body.md

-- Running vcpkg install - failed
CMake Error at dependencies/vcpkg/scripts/buildsystems/vcpkg.cmake:938 (message):
  vcpkg install failed.  See logs for more information:
  /Users/xxx/Documents/projects/Cemu/build/vcpkg-manifest-install.log
Call Stack (most recent call first):
  /usr/local/share/cmake/Modules/CMakeDetermineSystem.cmake:146 (include)
  CMakeLists.txt:56 (project)

-- Configuring incomplete, errors occurred!

install-x64-osx-dbg-out.log
---------------------------------------------------------------------------------------------------------
In file included from /Users/xxx/Documents/projects/Cemu/dependencies/vcpkg/buildtrees/hidapi/src/api-0.14.0-e052aaf522.clean/mac/hid.c:25:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDManager.h:29:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDLib.h:34:
/Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDLibObsolete.h:197:5: error: type name requires a specifier or qualifier
    IUNKNOWN_C_GUTS;
    ^
/Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDLibObsolete.h:204:5: error: type name requires a specifier or qualifier
    IUNKNOWN_C_GUTS;
    ^
/Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDLibObsolete.h:211:5: error: type name requires a specifier or qualifier
[... 49 lines cut ...]

typedef struct IOHIDDeviceInterface
{
    IUNKNOWN_C_GUTS;
    IOHIDDEVICEINTERFACE_FUNCS_100;
    IOHIDDEVICEINTERFACE_FUNCS_121;
} IOHIDDeviceInterface;

typedef struct IOHIDDeviceInterface121
{
    IUNKNOWN_C_GUTS;
    IOHIDDEVICEINTERFACE_FUNCS_100;
    IOHIDDEVICEINTERFACE_FUNCS_121;
} IOHIDDeviceInterface121;

........

---------------------------------------------------------------------------------------------------------
IUNKNOWN_C_GUTS
declares in /Library/Developer/CommandLineTools/SDKs/MacOSX14.5.sdk/System/Library/Frameworks/CoreFoundation.framework/Versions/A/Headers/CFPlugInCOM.h

---------------------------------------------------------------------------------------------------------
seems api-0.14.0-e052aaf522.clean/mac/hid.c lost necessary header 
#include <CoreFoundation/CFPlugInCOM.h>
or define COREFOUNDATION_CFPLUGINCOM_SEPARATE