libusb/hidapi #670: need longer timeout with send_feature_report() for libusb under Windows

oistra-ah, 2024-03-22:
Hi, could please someone give a hint: is it possible to increase timeout for send_feature_report() on Windows? I have the device, which takes a more than 5 sec sometimes for response. Thanks.

Youw, 2024-04-06:
As of this moment that is not possible.

Theoretically if could have been possible, but it would require 2 things:
1) Refactor `hid_send_feature_report` to switchi from `HidD_SetFeature` to using `DeviceIoControl` directly;
2) Introduce new API to specify the timeout.

> I have the device, which takes a more than 5 sec sometimes for response.

That is astucally weird from system design POV. If that is your device and you know for a fact that by setting a specific feature your device needs to perform a long-going operation, that is advisable not to perform that action directly during the `send_feature_report` handling, but rather immediately acknowledge that the operation has started, and send the result later in some async matter (e.g. using the interrupt input report).

Youw, 2024-04-06:
I any case I don't see much point keeping this ticket opened for now, unless someone volunteers to design the API and implement the change.

oistra-ah, 2024-06-02:
Hi @Youw , sorry for my late answer. And thanks for the reply.
1. I understood that this is possible but rather complex. My suspicions confirmed, thanks.
2. Sys design is weak, agree. And without waiting for your answer I've started to implement async handler)
3. Thus if nobody minds, you can close the ticket.

closed 2024-06-02 by Youw