libusb/hidapi #671: Warnings

Sod-Almighty, 2024-03-24:
[Warnings are errors](https://dev.solita.fi/2014/03/26/warnings-are-errors.html), people!

```
linux/hidapi/hid.c:23: warning: "_GNU_SOURCE" redefined
   23 | #define _GNU_SOURCE /* needed for wcsdup() before glibc 2.10 */
      |
<command-line>: note: this is the location of the previous definition
linux/hidapi/hid.c: In function ‘void read_callback(libusb_transfer*)’:
linux/hidapi/hid.c:953:37: error: invalid conversion from ‘void*’ to ‘hid_device*’ {aka ‘hid_device_*’} [-fpermissive]
  953 |         hid_device *dev = transfer->user_data;
      |                           ~~~~~~~~~~^~~~~~~~~
      |                                     |
      |                                     void*
linux/hidapi/hid.c: In function ‘void* read_thread(void*)’:
linux/hidapi/hid.c:1022:27: error: invalid conversion from ‘void*’ to ‘hid_device*’ {aka ‘hid_device_*’} [-fpermissive]
 1022 |         hid_device *dev = param;
      |                           ^~~~~
      |                           |
      |                           void*
linux/hidapi/hid.c: In function ‘void cleanup_mutex(void*)’:
linux/hidapi/hid.c:1480:27: error: invalid conversion from ‘void*’ to ‘hid_device*’ {aka ‘hid_device_*’} [-fpermissive]
 1480 |         hid_device *dev = param;
      |                           ^~~~~
      |                           |
      |                           void*
```

Youw, 2024-03-25:
PRs with fixes are welcome.

Youw, 2024-03-25:
> Because I have pride in my work.

And complete disrespect of other people's time and efforts.

---
I really appreciate reporting the problem - clearly our CI needs some improvements, I expected it [should have](https://github.com/libusb/hidapi/commit/52c728acf42b4cd86837074cb9d7cf646adc0d85) caught it.

> When someone reports a problem with my software

At some point, in order to support 10% of software I wrote at some point, I had to hire a team of 6 people.
If you're claiming you can support and fix all of the software you've ever written - it is either not much, or not much people is using it.