libusb/hidapi #673: Consern regarding Hotplug callbacks implementation

Youw, 2024-04-06:
A small concern here: calling any Register or Unregister functions from within a callback may (and probably will) cause a deadlock. On Windows, Critical Sections seem to keep an internal counter that allows the same thread to enter on multiple times, while in pthread, unless PTHREAD_MUTEX_RECURSIVE is specified, it will cause a deadlock.

Additionally, just unlocking the mutex before calling the callback (or making it recursive) will allow the user to remove elements in such a way that it disturbs the execution flow, which can already happen on Windows.

Bad scenarios:
* Removing the element right before the one currently being processed will invalidate the `current` pointer (as it will be pointing to an area that has just been `free`'d), and even if use-after-free situation doesn't break things, returning a value from the callback that deletes the callback will process the list incorrectly
* Removing the current element will lead to exactly the same use-after-free scenario
* Removing all elements may cause even more severe side-effects, as removing the last element triggers a cleanup sequence
* If the `next` pointer is stored in advance (which it isn't in current implementation, but should be warned against in the future), adding a new element while the last one is being processed will only cause the new one to not be processed while removing an element right after the current one will cause unpredictable behavior.

There is a way to circumvent this, by adding a flag to each callback that indicates that it should be removed later when it is considered safe, which is set by the Unregister function if the mutex is already locked at the moment of the call. It seems that, as long as we know for a fact that the mutex is locked by the same thread, it should be safe to read and navigate the list of callbacks and even set flags to it, and the list will not be changed (the mutex is still locked and will not be unloced until we leave the callback and the callback processing part). As for the Register function, I see no problem with additional callbacks being registered from a callback. However, this approach still requires to somehow know for a fact that the mutex is locked by the same thread we are currently in.

Another possible resolution would be to warn against the use of Register and Unregister functions inside a callback, however, there are scenarios where the user code could make use of that (a widely used example would be the scenario where the unplug callback is only registered after a device was connected).

_Originally posted by @k1-801 in https://github.com/libusb/hidapi/issues/238#issuecomment-2041012999_

k1-801, 2024-04-08:
Put a bit more thought into how that would work, the scenario where the current callback is removed might actually not cause any damage, but it will make the execution skip the next callback and immediately proceed to the one after the next one.

k1-801, 2025-05-18:
#676 was merged, so I assume this one is resolved

PR #676 (2024-09-06): Deadlock prevention patch for Connection-callback feature
This PR provides a solution to the possible issues that happen when a register or de-register call is made from inside a callback. The way it works is the same for all platforms and is described below.

Resolves #673 

1) The mutex is made recursive, so it can be locked by the same thread multiple times
2) ~~The `mutex_ready` variable is renamed to `mutex_state`, it is no longer a boolean value, and the value of 2 in it now indicates that the mutex is "In Use" (locked by a function that calls a callback).~~ `mutex_ready` kept intact, ad [... cut]
commit da500c6c (2024-09-06): Fix possible deadlock in Connection-callback feature (#676)

Fix the possible issues that happen when a register or de-register call is made from inside a callback. The way it works is the same for all platforms and is described below.

Resolves: #673 

1) The mutex is made recursive, so it can be locked by the same thread multiple times
2) `mutex_ready` kept intact, added 2 more flags, `mutex_in_use` and `cb_list_dirty`.
3) When the `mitex_in_use` flag is set, the Deregister call is no longer allowed to immediately remove any callbacks from the list. Instead, the `events` field in the callbac [... cut]
closed 2025-05-18 by Youw