libusb/hidapi #683: mistake in name

nitrologic, 2024-08-01:
https://github.com/libusb/hidapi/blame/4578ea24613af03e5631e400ed1167462702a112/mac/hid.c#L1344

I think this method should be 

hid_send_output_report

not

hid_send_output_feature_report

to match the public decl in hidapi.h[480]

mbcinergy, 2024-08-01:
You are correct, that must have been a copy/paste error. 

Strange that the macos build pipeline does not error on this.

Youw, 2024-08-02:
OK, I was planning on making a PR instead of pushing https://github.com/libusb/hidapi/commit/d101e5c7e4646ecde66b650b3b8ce4904acfb13a directly into master...

But since I didn't break anything (and, apparently fixed an issue) I will consider this time as an exception, and will add a protection rule to master branch to avoid such in the future.

Youw, 2024-08-02:
> Strange that the macos build pipeline does not error on this.

And, apparently, no one is using it yet.

commit d101e5c7 (2024-08-02): macOS: fix hid_send_output_report implementation

- fix name of the function on macOS
- add compile-time check by the CI

Fixes: #683
closed 2024-08-02 by Youw