libusb/hidapi #695: Windows: hid_write returns -1 on success

Megamouse, 2024-09-23:
On windows, hid_write returns -1 if WriteFile returns 1 (success).
function_result is only ever set to a different value if overlapped is true, which only happens if WriteFile returns 0.

This manifests in our log with `hid_write error: Success` e.g. with this pseudo-code:
```
if (hid_write(hidDevice, &report, sizeof(report)) == -1)
{
    std::cerr << "hid_write error: " << hid_error(hidDevice)) << std::endl;
}
```

Youw, 2024-09-25:
What version of HIDAPI are you on?

Megamouse, 2024-09-25:
0.14.0, but the relevant hid_write code didn't change on latest master either, so it should be present in the latest commit.

Youw, 2024-09-25:
> WriteFile returns 1

I imagined this shouldn't be possible once the `lpOverlapped` is set.
Is it not the case?

Megamouse, 2024-09-25:
Where do I check that?

Youw, 2024-09-26:
> WriteFile returns 1 

Well, if you're saying this is what you experience, I guess there is not much we can actually check here.
I guess to fix this, we need also pass the `lpNumberOfBytesWritten` into `WriteFile`, and in case if it returns `0` (SUCCESS) use `lpNumberOfBytesWritten` as a return value of the `hid_write`.

@Megamouse will you be able to implement/test this locally in your environment?

Megamouse, 2024-09-26:
I can try later.
In case you want to test this on your own, you need RPCS3 for windows, a DualShock3 controller and run the DualShock3 handler in the RPCS3 pad settings.
It's possible that we are doing something wrong with our implementation, but the code works, it's just that the error is logged (we only log it since the rumble/led isn't really important for running the software)

Megamouse, 2024-09-26:
This code change seems to fix the issue.
I'm not sure why it's not running async, but the `bytes_written` is equal to `length`

[image]

Youw, 2024-09-26:
Fire a PR, we'll have it reviewed and additionally tested.

PR #697 (2024-09-29): Windows: return bytes_written in hid_write if WriteFile returns 1 (succeeds synchronously)
Return bytes_written in hid_write if WriteFile returns 1, which means it succeeded synchronously.

According to [documentation](https://learn.microsoft.com/en-us/previous-versions/troubleshoot/windows/win32/asynchronous-disk-io-synchronous), the system reserves the right to run WriteFile synchronously, even if FILE_FLAG_OVERLAPPED is set.

"Be careful when you code for asynchronous I/O because the system reserves the right to make an operation synchronous if it needs to. So it is best if you write the pr [... cut]
closed 2024-09-29 by Youw