libusb/hidapi #703: `hid_get_report_descriptor`: missing unnumbered feature report

SuibianP, 2024-10-18:
On some devices, unnumbered feature reports seem omitted from the parsed `HIDP_PREPARSED_DATA`. The `hid_pp_caps_info` header indicates the count and size correctly, but marks it empty/unused.

```pike
# Feature caps_info struct:
pp_data->caps_info[2]->FirstCap           = 5
pp_data->caps_info[2]->LastCap            = 5
pp_data->caps_info[2]->NumberOfCaps       = 1
pp_data->caps_info[2]->ReportByteLength   = 91
```

```perl
# Output hid_pp_cap struct:
# Feature hid_pp_cap struct:
# Link Collections:
```

https://github.com/libusb/hidapi/blob/750bf201ae906eec1dc52c97087c8d0104c091ae/windows/hidapi_descriptor_reconstruct.h#L109-L114

If I anyhow print the trailing `phid_pp_cap` out, it consists entirely of zeros.

This is likely an existing bug/undocumented aspect of `HidD_GetPreparsedData` on MS side, and I am unsure if it could be worked around in userspace.

<details>
<summary>HID report descriptor (dumped on Linux with <code>hid-decode</code>)</summary>

```c
# device 0:0
# 0x05, 0x01,                    // Usage Page (Generic Desktop)        0
# 0x09, 0x02,                    // Usage (Mouse)                       2
# 0xa1, 0x01,                    // Collection (Application)            4
# 0x09, 0x01,                    //  Usage (Pointer)                    6
# 0xa1, 0x00,                    //  Collection (Physical)              8
# 0x05, 0x09,                    //   Usage Page (Button)               10
# 0x19, 0x01,                    //   Usage Minimum (1)                 12
# 0x29, 0x05,                    //   Usage Maximum (5)                 14
# 0x15, 0x00,                    //   Logical Minimum (0)               16
# 0x25, 0x01,                    //   Logical Maximum (1)               18
# 0x75, 0x01,                    //   Report Size (1)                   20
# 0x95, 0x05,                    //   Report Count (5)                  22
# 0x81, 0x02,                    //   Input (Data,Var,Abs)              24
# 0x75, 0x01,                    //   Report Size (1)                   26
# 0x95, 0x03,                    //   Report Count (3)                  28
# 0x81, 0x03,                    //   Input (Cnst,Var,Abs)              30
# 0x06, 0x00, 0xff,              //   Usage Page (Vendor Defined Page 1) 32
# 0x09, 0x40,                    //   Usage (Vendor Usage 0x40)         35
# 0x75, 0x08,                    //   Report Size (8)                   37
# 0x95, 0x02,                    //   Report Count (2)                  39
# 0x15, 0x81,                    //   Logical Minimum (-127)            41
# 0x25, 0x7f,                    //   Logical Maximum (127)             43
# 0x81, 0x02,                    //   Input (Data,Var,Abs)              45
# 0x05, 0x01,                    //   Usage Page (Generic Desktop)      47
# 0x09, 0x38,                    //   Usage (Wheel)                     49
# 0x15, 0x81,                    //   Logical Minimum (-127)            51
# 0x25, 0x7f,                    //   Logical Maximum (127)             53
# 0x75, 0x08,                    //   Report Size (8)                   55
# 0x95, 0x01,                    //   Report Count (1)                  57
# 0x81, 0x06,                    //   Input (Data,Var,Rel)              59
# 0x09, 0x30,                    //   Usage (X)                         61
# 0x09, 0x31,                    //   Usage (Y)                         63
# 0x16, 0x00, 0x80,              //   Logical Minimum (-32768)          65
# 0x26, 0xff, 0x7f,              //   Logical Maximum (32767)           68
# 0x75, 0x10,                    //   Report Size (16)                  71
# 0x95, 0x02,                    //   Report Count (2)                  73
# 0x81, 0x06,                    //   Input (Data,Var,Rel)              75
# 0xc0,                          //  End Collection                     77
# 0x06, 0x00, 0xff,              //  Usage Page (Vendor Defined Page 1) 78
# 0x09, 0x02,                    //  Usage (Vendor Usage 2)             81
# 0x15, 0x00,                    //  Logical Minimum (0)                83
# 0x25, 0x01,                    //  Logical Maximum (1)                85
# 0x75, 0x08,                    //  Report Size (8)                    87
[... 588 lines cut ...]
5F 73 74 72 69 6E 67 20 20 20 20 20 20 3D 20 22
52 61 7A 65 72 20 43 6F 62 72 61 22 0A 64 65 76
2D 3E 72 65 6C 65 61 73 65 5F 6E 75 6D 62 65 72
20 20 20 20 20 20 3D 20 30 78 30 32 30 30 0A 64
65 76 2D 3E 69 6E 74 65 72 66 61 63 65 5F 6E 75
6D 62 65 72 20 20 20 20 3D 20 30 0A 64 65 76 2D
3E 75 73 61 67 65 20 20 20 20 20 20 20 20 20 20
20 20 20 20 20 3D 20 30 78 30 30 30 32 0A 64 65
76 2D 3E 75 73 61 67 65 5F 70 61 67 65 20 20 20
20 20 20 20 20 20 20 3D 20 30 78 30 30 30 31 0A
64 65 76 2D 3E 70 61 74 68 20 20 20 20 20 20 20
20 20 20 20 20 20 20 20 20 3D 20 22 5C 5C 3F 5C
48 49 44 23 56 49 44 5F 31 35 33 32 26 50 49 44
5F 30 30 41 33 26 4D 49 5F 30 30 23 37 26 33 37
30 61 64 33 37 32 26 30 26 30 30 30 30 23 7B 34
64 31 65 35 35 62 32 2D 66 31 36 66 2D 31 31 63
66 2D 38 38 63 62 2D 30 30 31 31 31 31 30 30 30
30 33 30 7D 22 0A 0A 23 20 50 72 65 70 61 72 73
65 64 20 44 61 74 61 20 73 74 72 75 63 74 3A 0A
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

(Followed by all `0x00` until `HID_API_MAX_REPORT_DESCRIPTOR_SIZE`)

</details>

JoergAtGithub, 2024-10-19:
This is a known limitation of the reconstructor: All bits in the feature report are const, and Windows removes any information about const caps during the pre-parsing step. The reconstructor determines these const padding bits from the bit position of the next non-const cap, but this does not work for additional const bytes at the end of the report.

But your example shows that in this particular case there is the possibility, to improve this. While we have no information about additional const bytes for a particular report, Windows tells us the needed buffer size for input, output and feature reports. This is the size of the largest report + one byte. In case of devices that don't use ReportIDs, we know that there can be only one report of each type. So we could determine the number of additional const bytes at the end of the report from hid_pp_caps_info->ReportByteLength.

Youw, 2024-10-29:
@SuibianP would you be able to check if #707 brings any improvement for your case?

SuibianP, 2024-10-30:
@Youw Tried dumping from the device just now and `hid_report_reconstructor_test` adds the last const report correctly. Usage page and usage are gone for good as expected.

@JoergAtGithub Thanks for the implementation!

PR #707 (2024-10-30): Reconstruction logic for padding bytes at the end of Reports (only possible for devices without ReportID)
- Added reconstruction logic of const padding bytes at the end of a report for devices which don't use ReportIDs
- Enhanced padding logic to distinguish between byte and bit padding.
- Added a new test case for Razer Cobra mouse `1532_00A3_0002_0001` which has a completly con feature report of 90 bytes.
- Modified expected report descriptor reference file `045E_02FF_0005_0001_expected.rpt_desc` to include additional padding bytes.

Closes: #703
closed 2024-10-30 by Youw