libusb/hidapi #710: Cannot connect device to my program on Linux, but it works on Windows

luxdragon, 2024-11-27:
Greetings,

I'm using HIDAPI to write a cross-platform library for the FT260 USB-to-I2C bridge and have a problem. The code is available on Codeberg: https://codeberg.org/Zamenhof/FreeFT260

but basically, while the code compiles successfully on both Windows and Linux:
```
#define VID 0x0403 // Set the VID and PID of your bridge here
#define PID 0x6030

int main(void) {
    unsigned char i2c_hdc2010_addr = 0x40; // Set your address here
    unsigned char eeprom_addr = 0x54; // The EEPROM address
	wchar_t wstr[MAX_STR];

    if (hid_init()) {
        // We initialise the Hidapi backend
        fprintf(stderr, "Failed to initialise the hidapi-Library\n");
        return -1;
    }

    hid_device* device = hid_open(VID, PID, NULL);
    if (!device) {
        fprintf(stderr, "Device not openable: %ls\n", hid_error(NULL));
        hid_exit();
        return -1;
    }
```
It always returns ``` Device not openable: No HID devices with requested VID/PID found in the system.``` on Linux even though when I use lsusb the device is found and the VID and PID are correct. I ran the program both with sudo and without and it didn't make a change - according to lsusb my device is the only one with the VID/PID-combo. 

How come and how can I fix this issue?

mcuee, 2024-11-27:
What is the driver bound to the device? Is it hidraw or some other driver? hidapi under Linux only supports hidraw or libusb.

Can you try to detach the kernel hid driver to see if you can use hidapi libusb backend with your device?

luxdragon, 2024-11-27:
Heya thank you for your super fast response! Running lsusb gives the following output:
``` 
luxdragon@luxdragon:~/Documents/FreeFT260$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 006: ID 0403:6030 Future Technology Devices International, Ltd FT260
```

and using lsusb -t gives:
``` 
luxdragon@luxdragon:~/Documents/FreeFT260$ lsusb -t
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    |__ Port 2: Dev 6, If 0, Class=Human Interface Device, Driver=, 12M
    |__ Port 2: Dev 6, If 1, Class=Human Interface Device, Driver=usbhid, 12M
```

I tried detatching the module by running:
``` 
#include <libusb-1.0/libusb.h>
#include <stdio.h>

int main() {
    libusb_device_handle *handle;
    libusb_context *ctx = NULL;
    int r;

    r = libusb_init(&ctx);
    if (r < 0) {
        fprintf(stderr, "Failed to initialize libusb\n");
        return -1;
    }

    handle = libusb_open_device_with_vid_pid(ctx, 0x0403, 0x6030);
    if (!handle) {
        fprintf(stderr, "Cannot open device\n");
        libusb_exit(ctx);
        return -1;
    }

    if (libusb_kernel_driver_active(handle, 0)) {
        printf("Kernel driver active, detaching it...\n");
        r = libusb_detach_kernel_driver(handle, 0);
        if (r < 0) {
            fprintf(stderr, "Error detaching kernel driver: %s\n", libusb_strerror(r));
            libusb_close(handle);
            libusb_exit(ctx);
            return -1;
        }
    }

    printf("Kernel driver detached\n");
    libusb_close(handle);
    libusb_exit(ctx);
    return 0;
}
```
But it didn't fix anything

Youw, 2024-11-27:
> libusb_kernel_driver_active(handle, 0)

You've checked the interface `0` of the device, but it is clear from `lsusb -t` that `If 0` doesn't have any kernel driver active.

---
Try using `hid_enumerate` and see what devices do you have in your system at all.

Also, you may check manually if your device is present in the system at all, by: `ls -l /sys/class/hidraw/` and checking each hidraw device in the system, e.g. by `cat /sys/class/hidraw/hidraw0/device/uevent`

luxdragon, 2024-11-28:
Running ```  hid_enumerate ``` doesn't work, because the command can't be found, should I run the hid_enumerate from the repository.

Running ```  ls -l /sys/class/hidraw/``` gives:
```  totalo 0
lrwxrwxrwx 1 root root 0 Nov 28 07:34 hidraw0 -> ../../devices/platform/AMDI0010:01/i2c-0/i2c-ELAN1200:00/0018:04F3:30BA.0001/hidraw/hidraw0
lrwxrwxrwx 1 root root 0 Nov 28 07:34 hidraw1 -> ../../devices/pci0000:00/0000:00:08.1/0000:05:00.3/usb1/1-2/1-2:1.0/0003:046D:C534.0002/hidraw/hidraw1
lrwxrwxrwx 1 root root 0 Nov 28 07:34 hidraw2 -> ../../devices/pci0000:00/0000:00:08.1/0000:05:00.3/usb1/1-2/1-2:1.1/0003:046D:C534.0003/hidraw/hidraw2
lrwxrwxrwx 1 root root 0 Nov 28 07:34 hidraw3 -> ../../devices/pci0000:00/0000:00:08.1/0000:05:00.3/usb1/1-2/1-2:1.1/0003:046D:C534.0003/0003:046D:4054.0004/hidraw/hidraw3
```

And as for the last command:
``` luxdragon@luxdragon:~$ cat /sys/class/hidraw/hidraw0/device/uevent
DRIVER=hid-multitouch
HID_ID=0018:000004F3:000030BA
HID_NAME=ELAN1200:00 04F3:30BA
HID_PHYS=i2c-ELAN1200:00
HID_UNIQ=
MODALIAS=hid:b0018g0004v000004F3p000030BA```

Youw, 2024-12-05:
> And as for the last command:

I showed last command for _hidraw**0**_ as an example. Try it for your device, with specific _hidraw**X**_

mcuee, 2025-04-02:
Close this item since no activity for quite some time.

closed 2025-04-02 by mcuee