libusb/hidapi #711: How to distinguish between USB/BT for same device

MarkaRagnos0815, 2024-11-28:
I test a bit using PS4 Controller via USB and BT (Windows).
I am able to connect via USB and BT at the same time. 

the hid_device_info gives to different paths, except for the serial_number and interface_number i get teh same values.
For blutooth the serial number is empty. (I guess serial number is the same)

Is there any way to check if these two devices are the same?

JoergAtGithub, 2024-12-02:
On Windows you can use the the https://github.com/libusb/hidapi/blob/ff67c77daddbd8e61ad3873ac16f8edc005f943f/windows/hidapi_winapi.h#L39-L54

MarkaRagnos0815, 2024-12-02:
Thanks for sharing. Found a different solution by reading the Mac address from USB. Bt seems to have a serial number. the usb Mac address is the same as USB serial number. So I am able to find same device via vendor, product and serial number

Youw, 2024-12-03:
What about [`hid_bus_type`](https://github.com/libusb/hidapi/blob/ff67c77daddbd8e61ad3873ac16f8edc005f943f/hidapi/hidapi.h#L186) ? Should give you USB or BT.

mcuee, 2025-04-02:
Close this one since no updates for a while.

closed 2025-04-02 by mcuee

MarkaRagnos0815, 2025-04-02:
@Youw sorry for the late answer. The Problem is that the device shows up two times. One as USB and the other one as BT. But both device descriptors are completely different. My solution of reading the serial number in case of USB works quite fine. In this case the serial is identical.