libusb/hidapi #715: Hidapi does not recognize a device on a Mac M2 under Sonoma 14.7.2, but on Linux and Windows in Parallels

felias-fogg, 2025-01-22:
Hi,

I try to work with an Atmel-ICE device (hardware debugger), and it is not recognized on my M2 Mac under Sonoma 14.7.2. Running hidapitester and hidapi_enum.py gives very fishy results:

```

> hidapi_enum.py
------------------------------------------------------------
path:                "DevSrvsID:4295190229"
vendor_id:           0x004c
product_id:          0x0269
serial_number:       "4c-57-ca-eb-ff-c5"
release_number:      0x0000
manufacturer_string: "Apple Inc."
product_string:      "Magic Mouse"
usage_page:          65280
usage:               11
interface_number:    -1

------------------------------------------------------------
path:                "DevSrvsID:4295190229"
vendor_id:           0x004c
product_id:          0x0269
serial_number:       "4c-57-ca-eb-ff-c5"
release_number:      0x0000
manufacturer_string: "Apple Inc."
product_string:      "Magic Mouse"
usage_page:          65280
usage:               20
interface_number:    -1

------------------------------------------------------------
path:                "DevSrvsID:4294969535"
vendor_id:           0x0000
product_id:          0x0000
serial_number:       ""
release_number:      0x0000
manufacturer_string: "Apple"
product_string:      ""
usage_page:          65292
usage:               5
interface_number:    -1

------------------------------------------------------------
path:                "DevSrvsID:4295190243"
vendor_id:           0x004c
product_id:          0x0269
serial_number:       "4c-57-ca-eb-ff-c5"
release_number:      0x0000
manufacturer_string: "Apple Inc."
product_string:      "Magic Mouse"
usage_page:          65280
usage:               3
interface_number:    -1

------------------------------------------------------------
path:                "DevSrvsID:4295232238"
vendor_id:           0x05ac
product_id:          0x0221
serial_number:       ""
release_number:      0x0071
manufacturer_string: "Apple, Inc"
product_string:      "Apple Keyboard"
usage_page:          1
usage:               6
interface_number:    0

------------------------------------------------------------
path:                "DevSrvsID:4295190237"
vendor_id:           0x004c
product_id:          0x0269
serial_number:       "4c-57-ca-eb-ff-c5"
release_number:      0x0000
manufacturer_string: "Apple Inc."
product_string:      "Magic Mouse"
usage_page:          1
usage:               2
interface_number:    -1

------------------------------------------------------------
path:                "DevSrvsID:4295190237"
vendor_id:           0x004c
product_id:          0x0269
serial_number:       "4c-57-ca-eb-ff-c5"
release_number:      0x0000
manufacturer_string: "Apple Inc."
product_string:      "Magic Mouse"
usage_page:          1
usage:               1
interface_number:    -1

------------------------------------------------------------
path:                "DevSrvsID:4295190237"
vendor_id:           0x004c
product_id:          0x0269
serial_number:       "4c-57-ca-eb-ff-c5"
release_number:      0x0000
manufacturer_string: "Apple Inc."
product_string:      "Magic Mouse"
usage_page:          13
usage:               5
interface_number:    -1

------------------------------------------------------------
path:                "DevSrvsID:4295190237"
vendor_id:           0x004c
product_id:          0x0269
serial_number:       "4c-57-ca-eb-ff-c5"
release_number:      0x0000
manufacturer_string: "Apple Inc."
product_string:      "Magic Mouse"
usage_page:          65280
usage:               12
interface_number:    -1

------------------------------------------------------------
path:                "DevSrvsID:4294969376"
vendor_id:           0x0000
product_id:          0x0000
serial_number:       ""
release_number:      0x0000
manufacturer_string: "APPL"
product_string:      "BTM"
usage_page:          65280
usage:               72
interface_number:    -1

------------------------------------------------------------
path:                "DevSrvsID:4295190222"
vendor_id:           0x004c
product_id:          0x0269
serial_number:       "4c-57-ca-eb-ff-c5"
release_number:      0x0311
manufacturer_string: "Apple Inc."
product_string:      "Magic Mouse"
usage_page:          0
usage:               1
interface_number:    -1

------------------------------------------------------------
path:                "DevSrvsID:4294969534"
vendor_id:           0x0000
product_id:          0x0000
serial_number:       ""
release_number:      0x0000
manufacturer_string: "Apple"
product_string:      ""
usage_page:          65280
usage:               255
interface_number:    -1

------------------------------------------------------------
path:                "DevSrvsID:4295221964"
vendor_id:           0x056d
product_id:          0x4065
serial_number:       ""
release_number:      0x8000
manufacturer_string: "EIZO corporation"
product_string:      "EIZO USB HID Monitor"
usage_page:          128
usage:               1
interface_number:    0

------------------------------------------------------------
path:                "DevSrvsID:4294968858"
vendor_id:           0x0000
product_id:          0x0000
serial_number:       ""
release_number:      0x0000
manufacturer_string: "Apple"
product_string:      "Headset"
usage_page:          12
usage:               1
interface_number:    -1

------------------------------------------------------------
path:                "DevSrvsID:4295232242"
vendor_id:           0x05ac
product_id:          0x0221
serial_number:       ""
release_number:      0x0071
manufacturer_string: "Apple, Inc"
product_string:      "Apple Keyboard"
usage_page:          12
usage:               1
interface_number:    1

------------------------------------------------------------
path:                "DevSrvsID:4294971079"
vendor_id:           0x046d
product_id:          0x085e
serial_number:       "D1C08FAB"
release_number:      0x0317
manufacturer_string: ""
product_string:      "Logitech BRIO"
usage_page:          12
usage:               1
interface_number:    5

------------------------------------------------------------
path:                "DevSrvsID:4294969531"
vendor_id:           0x0000
product_id:          0x0000
serial_number:       ""
release_number:      0x0000
manufacturer_string: "Apple"
product_string:      ""
usage_page:          65292
usage:               1
interface_number:    -1

```
So, no ATMEL-ICE, but a lot of devices with a VIDPID of 0:0. BTW, the correct entry (recorded under Ubuntu 24.04 (ARM64) run in the Paralleles Emulator) looks like that:

```
path:                "1-1:1.0"
vendor_id:           0x03eb
product_id:          0x2141
serial_number:       "J42700018184"
release_number:      0x0101
manufacturer_string: "Atmel Corp."
product_string:      "Atmel-ICE CMSIS-DAP"
usage_page:          0
usage:               0
interface_number:    0
```

Also under Windows11(ARM), and on my Intel-Mac, everything works. So, I am at loss here.

Cheers,
Bernhard

Youw, 2025-01-23:
> but a lot of devices with a VIDPID of 0:0. BTW

The manufacturer is Apple, and the bus is probably not USB, so that's expected on macOS.

> (recorded under Ubuntu 24.04 (ARM64) run in the Paralleles Emulator)

Is it run on that mac machine?

Maybe you've passed the USB device into you VM env and the host USB/HID driver gets disabled?

What is the output of `ioreg -p IOUSB ` or `ioreg -p IOUSB -w0 -l` on your mac?

felias-fogg, 2025-01-23:
Hi,

> 
> > (recorded under Ubuntu 24.04 (ARM64) run in the Paralleles Emulator)
> 
> Is it run on that mac machine?

Yes. Same machine. 

> 
> Maybe you've passed the USB device into you VM env and the host USB/HID driver gets disabled?

Oops. Nah, I disabled the USB access before, of course. What is more, a lot of programs can access the Atmel-ICE, e.g., avrdude and MPLABX.

> 
> What is the output of `ioreg -p IOUSB ` or `ioreg -p IOUSB -w0 -l` on your mac?

Here we go (and the Atmel-ICE is enumerated):

```
   |
   |
  +-o AppleEmbeddedUSBXHCIASMedia3142@08000000  <class AppleEmbeddedUSBXHCIASMedia3142, id 0x1000008fb, registered, matched, active, busy 0 (4001 ms), retain 146>
    | {
    |   "IOClass" = "AppleEmbeddedUSBXHCIASMedia3142"
    |   "kUSBSleepPortCurrentLimit" = 2100
    |   "UsbHostControllerSoftRetryPolicy" = 0
    |   "IOPersonalityPublisher" = "com.apple.driver.usb.AppleEmbeddedUSBXHCIPCI"
    |   "IOMatchedAtBoot" = Yes
    |   "IOPowerManagement" = {"ChildrenPowerState"=3,"DevicePowerState"=2,"CurrentPowerState"=3,"CapabilityFlags"=32768,"MaxPowerState"=3,"DriverPowerState"=0}
    |   "IOProviderClass" = "IOPCIDevice"
    |   "kUSBWakePowerSupply" = 3100
    |   "IOProbeScore" = 10000
    |   "IOUserClientClass" = "AppleASMedia3142USBXHCIUserClient"
    |   "locationID" = 134217728
    |   "UsbRTD3Supported" = Yes
    |   "kUSBWakePortCurrentLimit" = 2100
    |   "IONameMatch" = "usb-xhci,ASM3142"
    |   "CFBundleIdentifierKernel" = "com.apple.driver.usb.AppleEmbeddedUSBXHCIPCI"
    |   "kUSBSleepPowerSupply" = 3100
    |   "IOMatchCategory" = "IODefaultMatchCategory"
    |   "CFBundleIdentifier" = "com.apple.driver.usb.AppleEmbeddedUSBXHCIPCI"
    |   "Revision" = <0103>
    |   "IONameMatched" = "usb-xhci,ASM3142"
    |   "UsbHostControllerUSB4LPMPolicy" = 1
    |   "UsbHostControllerTierLimit" = 6
    |   "UsbBusCurrentPoolID" = 4294969595
    |   "controller-statistics" = {"kControllerStatIOCount"=120703,"kControllerStatPowerStateTime"={"kPowerStateOff"="768ms (0%)","kPowerStateSleep"="39061981ms (11%)","kPowerStateOn"="287372591ms (88%)","kPowerStateSuspended"="3460ms (0%)"},"kControllerStatSpuriousInterruptCount"=0}
    |   "kUSBSleepSupported" = Yes
    | }
    | 
    +-o Keyboard Hub@08300000  <class IOUSBHostDevice, id 0x100040ad6, registered, matched, active, busy 0 (84 ms), retain 231>
    | | {
    | |   "kUSBSerialNumberString" = "000000000000"
    | |   "bDeviceClass" = 9
    | |   "bDeviceSubClass" = 0
    | |   "iSerialNumber" = 3
    | |   "IOServiceDEXTEntitlements" = (("com.apple.developer.driverkit.transport.usb"))
    | |   "iProduct" = 2
    | |   "USB Serial Number" = "000000000000"
    | |   "USB Vendor Name" = "Apple, Inc."
    | |   "USBSpeed" = 3
    | |   "IOPowerManagement" = {"PowerOverrideOn"=Yes,"CapabilityFlags"=32768,"MaxPowerState"=2,"DevicePowerState"=2,"DriverPowerState"=0,"ChildrenPowerState"=2,"CurrentPowerState"=2}
    | |   "bNumConfigurations" = 1
    | |   "kUSBProductString" = "Keyboard Hub"
    | |   "kUSBConfigurationCurrentOverride" = 500
    | |   "kUSBVendorString" = "Apple, Inc."
    | |   "USB Product Name" = "Keyboard Hub"
    | |   "iManufacturer" = 1
    | |   "idVendor" = 1452
[... 67 lines cut ...]
          "iSerialNumber" = 3
          "bNumConfigurations" = 1
          "UsbDeviceSignature" = <eb03412101014a3432373030303138313834ef0201030000ffffff>
          "USB Product Name" = "Atmel-ICE CMSIS-DAP"
          "locationID" = 138412032
          "bDeviceSubClass" = 2
          "bcdUSB" = 512
          "kUSBSerialNumberString" = "J42700018184"
          "USB Address" = 12
          "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
          "kUSBCurrentConfiguration" = 1
          "bDeviceProtocol" = 1
          "USBPortType" = 0
          "IOServiceDEXTEntitlements" = (("com.apple.developer.driverkit.transport.usb"))
          "USB Vendor Name" = "Atmel Corp."
          "Device Speed" = 2
          "idVendor" = 1003
          "kUSBProductString" = "Atmel-ICE CMSIS-DAP"
          "USB Serial Number" = "J42700018184"
          "IOGeneralInterest" = "IOCommand is not serializable"
          "kUSBAddress" = 12
          "kUSBVendorString" = "Atmel Corp."
        }

```

mcuee, 2025-01-24:
Hmm, this is very strange. Do you have other Macs to check whether this is specific to Sonoma 14 or with M2.

Atmel ICE is supported by avrdude under Linux/Windows/macOS/FreeBSD/etc, using hidapi. There is a bug with TPI parts but I have not heard issues like what you have mentioned here.

Reference:
* https://github.com/avrdudes/avrdude/discussions/1412

mcuee, 2025-01-24:
BTW, is it possible for you to upgrade your macOS version to the latest 15.2 verion and try again.

Apple is not really that good in terms of legacy USB device support compared to Windows and Linux and often there are breaking changes from time to time. But often the bug got fixed in later release.

felias-fogg, 2025-01-24:
I upgraded on my Intel Mac from Ventura to Sonoma, and the Atmel-ICE is still listed when using hidapitester or hidutil. So, it does not seem to be an OS problem.

I'll try upgrading to 15.2. on my ARM Mac hoping that the problem vanishes.

felias-fogg, 2025-01-24:
Hi,

The upgrade to 15.2 on my ARM Mac did not help.

felias-fogg, 2025-01-24:
It is a mystery to me, why the enumeration of hid devices leaves out the Atmel-ICE only on ARM Macs. However, I guess, I found a way around it (using what avrdude apparently does). If the hidapi enumeration fails, it goes through all USB devices and looks for the right VID:PID pair.

felias-fogg, 2025-01-29:
Turned out that it was just an issue of an old extension that grabbed the device.

closed 2025-01-29 by felias-fogg

Youw, 2025-01-29:
> Turned out that it was just an issue of an old extension that grabbed the device.

Sounds about right. Thank's for letting us know.