libusb/hidapi #717: Single source file building

toxieainc, 2025-02-25:
Hi!

https://github.com/libusb/hidapi/blob/master/BUILD.md#embedding-hidapi-directly-into-your-source-tree mentions that including hidapi into a repo for compiling should work via copying a single file <backend>/hid.c .
I tried for windows, but from what i experienced, i needed to add all .h and .c files from the windows folder.
Am i missing something or is this documentation simply outdated?

Thanks!

Youw, 2025-02-25:
> mentions that including hidapi into a repo for compiling should work via copying a single file /hid.c

Unfortunately you've misread the documentation. In fact:

1) you need to include a single source file into your project's build system
2) NOTE: the above **doesn't guarantee** that having a copy of <backend>/hid.c and hidapi/hidapi.h is enough to build HIDAPI. The only guarantee that _<backend>/hid.c includes all necessary sources to compile it as a single file_.

toxieainc, 2025-02-25:
Ah, okay, yup i misinterpreted the doc then. Its a bit unusual, but okay, thanks..

closed 2025-02-25 by toxieainc