libusb/hidapi #720: hidapi, hid_write, and resume from suspend

netfab, 2025-03-01:
Hi,

I wrote a linux daemon that performs USB transfers using hidapi hid_write().
Everything is running fine.

Today I started using suspend/resume feature on my system, and I noticed that when resuming from suspend, the daemon is producing a unique error :

> Mar  1 07:53:15 coreI5 kernel: Disabling non-boot CPUs ...
> Mar  1 07:53:15 coreI5 GLogiKd[2814]: hid_write error
> Mar  1 07:53:15 coreI5 GLogiKd[2814]: HIDAPI error : hid_error is not implemented yet
> Mar  1 07:53:15 coreI5 GLogiKd[2814]: LCD refresh failure
> Mar  1 07:53:15 coreI5 e-suspend[11429]: System returned from sleep operation 'suspend'.

After that, everything is running fine, and the deamon is resuming perfectly.

My guess is that suspend/resume operations interrupts the blocking libusb_interrupt_transfer() call, and then [hid_write() returns -1](https://github.com/libusb/hidapi/blob/master/libusb/hid.c#L1445) on resume  :
```
if (res < 0)
	return -1;
```

From the [libusb documentation](https://libusb.sourceforge.io/api-1.0/group__libusb__syncio.html#ga0f171a72904a552fc43e6e6564d108a3) we can find libusb_interrupt_transfer() return values :

> Returns
>    0 on success (and populates transferred) 
>     LIBUSB_ERROR_TIMEOUT if the transfer timed out 
>     LIBUSB_ERROR_PIPE if the endpoint halted 
>     LIBUSB_ERROR_OVERFLOW if the device offered more data, see Packets and overflows 
>     LIBUSB_ERROR_NO_DEVICE if the device has been disconnected 
>     LIBUSB_ERROR_BUSY if called from event handling context 
>     LIBUSB_ERROR_INVALID_PARAM if the transfer size is larger than the operating system and/or
>       hardware can support (see Transfer length limitations) another LIBUSB_ERROR code
>       on other error 

Here is my question : would you consider (instead of simply returning -1) to return the real return value from libusb ?

```
if (res < 0)
	return res;
```

This would allows me to handle hid_write() errors with better granularity. In this case, not really an error, probably a unique timeout on each resume ?

Thank you.

netfab, 2025-03-01:
Obviously I realize that this may create problems for other implementations.

Youw, 2025-03-01:
Take a look at: https://github.com/libusb/hidapi/pull/708/files (newly added `hid_libusb_error` function)

Might be just what you need.

netfab, 2025-03-01:
Yes I think it will do the job. I will fork and test that ASAP.
For now I'm closing this issue, if something goes wrong I will reopen.
Thank you.

closed 2025-03-01 by netfab

Youw, 2025-03-01:
Make sure to note: https://github.com/libusb/hidapi/wiki/Multi%E2%80%90threading-Notes

That is the branch where hid_error is implemented for libusb backend as well.

netfab, 2025-03-21:
For the record, I discovered that when resuming from hibernate, the kernel is automatically resetting each connected USB device on the system :
> Mar 21 13:03:57 coreI5 e-hibernate[15709]: System returned from sleep operation 'hibernate'.
> Mar 21 13:03:57 coreI5 kernel: Restarting tasks ... done.
> Mar 21 13:03:57 coreI5 kernel: PM: hibernation: hibernation exit
> [...]
> Mar 21 13:03:57 coreI5 kernel: usb 2-1.3: reset high-speed USB device number 3 using ehci-pci
> Mar 21 13:03:57 coreI5 kernel: usb 1-1.6: reset low-speed USB device number 3 using ehci-pci
> Mar 21 13:03:57 coreI5 kernel: usb 3-2: reset high-speed USB device number 6 using xhci_hcd
> Mar 21 13:03:57 coreI5 kernel: usb 3-2.2: reset full-speed USB device number 7 using xhci_hcd
> Mar 21 13:03:57 coreI5 kernel: usb 3-2.3: reset full-speed USB device number 8 using xhci_hcd
> Mar 21 13:03:57 coreI5 kernel: usb 3-2.4: reset full-speed USB device number 9 using xhci_hcd

This means that each opened USB connection must be closed **before** entering sleep state, and reopened **after** resume. On linux, this can be done by catching the *PrepareForSleep* DBus signal (on the system bus) sent by systemd-logind (see link 1).

So, finally, the initial error is workarounded, and I do not need the return value from libusb.

1. https://www.freedesktop.org/wiki/Software/systemd/inhibit/