libusb/hidapi #724: Fix netbsd/freebsd/openbsd CI

Youw, 2025-03-03:
>  failed to submit sr.ht job: gqlclient: HTTP server error (403 Forbidden)

z3ntu, 2025-03-04:
Yeah, it should still work. Maybe it needs to be re-authorized due to the bridge losing its refresh token or something?

Youw, 2025-03-04:
You probably should have enough access to do so yourself.
If not - let me know what/where to check.

Youw, 2025-03-04:
New installation ID: https://github.com/organizations/libusb/settings/installations/62055409

z3ntu, 2025-03-04:
Like last time I went to https://hottub.emersion.fr/post-install?installation_id=62055409

It says now installation is complete. Can you try if it works again?

Youw, 2025-03-04:
All good, thanks!

closed 2025-03-04 by Youw

z3ntu, 2025-03-04:
Awesome, good to hear! 

So in case it happens again, should be two steps essentially, one for you and one for me :)