libusb/hidapi #729: pp_data_dump issue for 046A_0011_0006_0001

mcuee, 2025-03-27:
There is a typo here.

https://github.com/libusb/hidapi/blob/0ab6c14264ec76e9328a8eedb3b72b5d27dffd47/windows/test/data/046A_0011_0006_0001.pp_data#L4
```
dev->manufacturer_string = "dev->product_string      = "dev->release_number      = 0x0100
```

* https://github.com/libusb/hidapi/pull/728#issuecomment-2756866533

>There is something missing, as this is autogenerated by `pp_data_dump`, I guess there was something in the manufacturer_string , that `pp_data_dump` couldn't handle. Could you please open a dedicated issue for this, as this is unrelated to this PR.

JoergAtGithub, 2025-03-27:
@mcuee Is this one of your devices? Without the device, we are neither able to restore the missing information, nor can we debug what was going wrong.

mcuee, 2025-03-28:
@JoergAtGithub 

Google says it is a "Cherry G83 (RS 6000) Keyboard" so it is not my device.
https://devicehunt.com/view/type/usb/vendor/046A/device/0011

mcuee, 2025-03-28:
@JoergAtGithub 

Still could it be a accident change to the file?
```
dev->manufacturer_string = ""
dev->product_string      = ""
dev->release_number      = 0x0100 
```

mcuee, 2025-03-28:
Here is the zip file for one system (Dell Laptop, Windows 11 Enterprise 23H2). I do not see any issues with this system. I will try to check my other Windows systems.

The zip file contains everything: Windows x64 binary, the two generated txt files and 27 generated pp_data file.

```
PS C:\work\libusb\hidapi\build_msvc\bin_x64\RelWithDebInfo> .\pp_data_dump.exe > pp_data_dump_dell_laptop.txt
PS C:\work\libusb\hidapi\build_msvc\bin_x64\RelWithDebInfo> .\hidtest.exe >hidtest_dell_laptop.txt
```

[test_data_dell_laptop.zip](https://github.com/user-attachments/files/19499441/test_data_dell_laptop.zip)

mcuee, 2025-03-29:
Here is the zip file for another system (Acer Swift 3 Laptop early 2021 model, Windows 11 Home 24H2). I do not see any obvious issues with this system. I will try to check my other Windows systems.

The zip file contains everything: Windows x64 binary, the two generated txt files and the generated pp_data files.

[test_data_acer_swift_3_2021.zip](https://github.com/user-attachments/files/19521324/test_data_acer_swift_3_2021.zip)

mcuee, 2025-04-19:
> But I will still check my devices to see if my no-brand USB Mechanical Keyboard uses the same VID/PID.

Unfortunately I do not own the device 046A:0001.

mcuee, 2025-04-19:
Since we cannot do anything about this issue now, I will close this issue for now.

closed 2025-04-19 by mcuee

mcuee, 2025-04-20:
Here is the zip file for another system (Acer Swift Go 14 Laptop 2024 model, Windows 11 Home 24H2). I do not see any obvious issues with this system.

The zip file contains everything: Windows x64 binary, the two generated txt files and the generated pp_data files.

[acer_swift_go_14_2024.zip](https://github.com/user-attachments/files/19825193/acer_swift_go_14_2024.zip)