libusb/hidapi #732: arch Linux github action build failed

mcuee, 2025-04-03:
Reference:
https://github.com/libusb/hidapi/actions/runs/14176013837/job/39733910021?pr=730
https://github.com/libusb/hidapi/actions/runs/14115441113/job/39887008896?pr=728

mcuee, 2025-04-03:
Maybe not a hidapi issue but rather issues of PR #728 and PR #730 though.

mcuee, 2025-04-03:
But the failure in PR #730 seems to be strange.

```
Run cmake \
-- The C compiler identification is GNU 14.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/sbin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
CMake Error at CMakeLists.txt:12 (find_package):
  Could not find a configuration file for package "hidapi" that is compatible
  with requested version "0.12".

  The following configuration files were considered but not accepted:

    install/shared-cmake/lib/cmake/hidapi/hidapi-config.cmake, version: unknown

-- Configuring incomplete, errors occurred!
```

Youw, 2025-04-03:
Right - both are PR issues, not repo issues, to be fixed in scope of those PRs by their authors.

aokblast, 2025-04-03:
I see the artifact build 0.15.0 (  /__w/hidapi/hidapi/install/shared-cmake/lib/libhidapi-hidraw.so.0.15.0).
But what records in the CMakeLists.txt file in test program is 0.12.0

Youw, 2025-04-03:
That shouldn't be a problem, as CMake would look for same Major version and same or newer Minor version - so it is perfectly fine to require `0.12` and get `0.15.0`.

aokblast, 2025-04-03:
Emm... I create a archlinux vm with main branch(without my code). I also cannot pass the check when repeat all the CI command .

Youw, 2025-04-03:
> I use absolute path for hidapi_ROOT and it works.

Could you be more specific?

Youw, 2025-04-04:
Nope, Archlinux is a CI/GithubActions issue - master is failing as well. Will check.,

PR #733 (2025-04-04): Fix Archlinux CI build
Fix CI build on latest Archlinux with CMake 4.0

Fixes: #732
closed 2025-04-04 by Youw