libusb/hidapi #735: Cannot compile hidapi 1.4.0 on macOS 15.4.1 with Xcode 16.x

iamqk, 2025-04-23:
Encounter similar error and Reply at https://github.com/libusb/hidapi/issues/669

Youw, 2025-04-24:
This is weird. HIDAPI doesn't use CFPlugin directly, so it shouldn't be included in hid.c.
@iamqk could you debug a bit and check is `COREFOUNDATION_CFPLUGINCOM_SEPARATE` not defined at all, or defined to `0`?

iamqk, 2025-04-25:
> This is weird. HIDAPI doesn't use CFPlugin directly, so it shouldn't be included in hid.c.

I update brew llvm to @17, the error goes away
(but directly compile hidtest using llvm17 still have some error:

 make -f Makefile-manual                        
/usr/local/opt/llvm@17/bin/clang++ -I../hidapi -I. -Wall -g -c hid.c -o hid.o
clang++: warning: treating 'c' input as 'c++' when in C++ mode, this behavior is deprecated [-Wdeprecated]
hid.c:342:13: error: no matching function for call to 'CFNumberGetValue'
  342 |                         result = CFNumberGetValue(ref, kCFNumberSInt32Type, out_val);
      |                                  ^~~~~~~~~~~~~~~~
/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/System/Library/Frameworks/CoreFoundation.framework/Headers/CFNumber.h:105:9: note: candidate function not viable: cannot convert argument of incomplete type 'CFTypeRef' (aka 'const void *') to 'CFNumberRef' (aka 'const __CFNumber *') for 1st argument
  105 | Boolean CFNumberGetValue(CFNumberRef number, CFNumberType theType, void *valuePtr);
      |         ^                ~~~~~~~~~~~~~~~~~~
hid.c:589:19: error: assigning to 'char *' from incompatible type 'void *'
  589 |                 cur_dev->path = calloc(1, path_len);
      |                                 ^~~~~~~~~~~~~~~~~~~
2 errors generated.

), make through.

Thx for the tips.

closed 2025-04-25 by iamqk