libusb/hidapi #744: HIDAPI does not recognize "virtual" HID device

tuxedo-wse, 2025-05-21:
So with this driver https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/?id=cfd84b3f419bf0aec60ecddc92c61b539c339ec9 there is a virtual HID device i.e. a hid device that exists just in software and is not connected to a bus.

HIDAPI does net seems to recognize this.

My minimal test code:
```
#include <stdio.h>
#include <hidapi/hidapi.h>

int main() {
    printf("HID Devices:\n");
    struct hid_device_info *devi, *dev = hid_enumerate(0, 0);
    devi = dev;
    while(devi != NULL) {
        printf("%04x:%04x\n", devi->vendor_id, devi->product_id);
        devi = devi->next;
    }
    hid_free_enumeration(dev);
}
```
Compiled with `gcc test.c -lhidapi-hidraw` for the hidraw interface, this does not liste the device, while /sys/class/hidraw/ has the device.

mcuee, 2025-05-21:
Does this device show up as /dev/hidraw*? If this device is not a USB hid device and yet not show up as hidraw device, then hidapi cannot find it.

mcuee, 2025-05-21:
Basically you can run the Linux kernel hid-example to see if that works or not.
https://github.com/torvalds/linux/blob/master/samples/hidraw/hid-example.c

tuxedo-wse, 2025-05-21:
> Does this device show up as /dev/hidraw*?

Yes

> Also you need to have permission. You may want to try to run as sudo to see if that helps.

I did that but didn't help

tuxedo-wse, 2025-05-21:
More testing using hid_open_path:

```
#include <stdio.h>

#include <hidapi/hidapi.h>

int main() {
    printf("HID Devices:\n");
    struct hid_device_info *devi, *dev = hid_enumerate(0, 0);
    devi = dev;
    while(devi != NULL) {
        printf("%04x:%04x\n", devi->vendor_id, devi->product_id);
        devi = devi->next;
    }
    hid_free_enumeration(dev);

    hid_device *foo = hid_open_path("/dev/hidraw2");
    devi = hid_get_device_info(foo);
    printf("\n%04x:%04x\n", devi->vendor_id, devi->product_id);
    hid_close(foo);

    foo = hid_open_path("/dev/hidraw3");
    devi = hid_get_device_info(foo);
    printf("\n%04x:%04x\n", devi->vendor_id, devi->product_id);
    hid_close(foo);
}
```
Both /dev/hidraw2 and /dev/hidraw3 exist and i run the programm with root.

hidraw2 is a normal physical device (I think the usb keyboard), hidraw3 is the virtual device from the driver.

The hidraw 2 example works fine while the hidraw3 example crashes as the printf step

tuxedo-wse, 2025-05-21:
Following that trace my educated guess is that the culprit is this switch statement: https://github.com/libusb/hidapi/blob/d6b2a974608dec3b76fb1e36c189f22b9cf3650c/linux/hid.c#L716

It expects the bus type to be USB, BT, I2C, or SPI, but this device is none of that (in the kernel driver it is specified it as BUS_VIRTUAL https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/tree/drivers/platform/x86/tuxedo/nb04/wmi_ab.c?id=cfd84b3f419bf0aec60ecddc92c61b539c339ec9#n841 , I guess that is the same define that is asked for here?)

tuxedo-wse, 2025-05-21:
there is a switch statement before that with a clear comment: https://github.com/libusb/hidapi/blob/d6b2a974608dec3b76fb1e36c189f22b9cf3650c/linux/hid.c#L680

closed 2025-05-23 by Youw