libusb/hidapi #752: workarounds for bad report descriptors?

greyltc, 2025-06-15:
I've got a device with a firmware bug that means it returns a malformed report descriptor. In practice, this means that in Linux the hid-generic driver gives up while trying to bind to the device with
```
[ 2.258067] hid-generic 0003:3746:FFFF.0001: unbalanced collection at end of report description
[ 2.258092] hid-generic 0003:3746:FFFF.0001: probe with driver hid-generic failed with error -22
```
eg. https://github.com/networkupstools/nut/issues/2735#issuecomment-2901386668

In Windows, the built-in hid driver unsurprisingly pukes too; I get a little exclamation mark in device manager with an error message like
```
This device cannot start. (Code 10)

Extra end collection found or end collection not found.
```
eg.

[image]

After some digging, it seems like the device leaves off three `0xC0` (end collection indicator) bytes from its report descriptor.

I've figured out how to address this in Linux: https://github.com/greyltc/r3pcomms/blob/8bd4e65655d546fb53ddde6b30d8380a4e53cb99/hid_fixes/r3p_rd_patch.c#L23-L29
and the hid-generic driver starts working again and /dev/hidrawX appears and I can happily use hidapi.

Do you know any workarounds I can apply in Windows to fix up the malformed report descriptor so that hidapi will work there too?

mcuee, 2025-06-16:
@greyltc 

Unfortunately this is out of the topic for HIDAPI project. 

If Windows USB Stack does not recognize the device, HIDAPI cannot help at all.

There is probably a way to fix the issue using a Windows Kernel driver, probably a lower filter driver to fix the report descriptor. But Windows kernel driver development is a very complex topic.

I will recommend you to post your question in OSR forum and mention this link. Tim Roberts is active in that forum and he is a Windows driver expert who is also familiar with libusb and hidapi.
https://community.osr.com/c/ntdev/7

I usually post the question about Windows USB and HID devices in that forum as well.

closed 2025-06-16 by mcuee

mcuee, 2025-06-16:
Similar question asked in OSR NTDEV forum.
https://community.osr.com/t/kmdf-usb-filter-driver-intercepting-interrupts-from-lower-driver/58555/3

greyltc, 2025-06-16:
Thanks for pointing me over there!

greyltc, 2025-06-16:
for posterity, here's the post I made over there about the same thing: https://community.osr.com/t/workarounds-for-bad-report-descriptors/59724