libusb/hidapi #755: add the call to the CancelIo function

jianweihao, 2025-06-24:
https://github.com/libusb/hidapi/blob/f18d2c0768bed300d13758dd9f52b57163c0498f/windows/hid.c#L1129C3-L1133C4

This piece of if-judgment code needs to add the call to the CancelIo function. Otherwise, after WaitForSingleObject times out and returns, WriteFile may fail with a certain probability and keep being stuck in the IO_PENDING state. It is necessary to add the CancelIo operation to continue sending data normally.

eg :
`
		if (res != WAIT_OBJECT_0) {
			/* There was a Timeout. */
			CancelIo(dev->device_handle); 
			register_winapi_error(dev, L"hid_write/WaitForSingleObject");
			goto end_of_function;
		}
`

Youw, 2025-06-24:
@jianweihao how did you end up checking that piece of code? Did you happen to reproduce it?

And there is a problem adding `CancelIo` before `register_winapi_error`...

mcuee, 2025-06-24:
> Yes, we have the logic for switching USB HID devices, which triggers USB plug-in and removal during use. This is when the issue occurs. After calling the CancelIo function, the problem is resolved.

Can you try this hotplug branch? Thank.
https://github.com/libusb/hidapi/tree/connection-callback

jianweihao, 2025-06-24:
> > Yes, we have the logic for switching USB HID devices, which triggers USB plug-in and removal during use. This is when the issue occurs. After calling the CancelIo function, the problem is resolved.
> 
> Can you try this hotplug branch? Thank. The current main branch does not really support hotplug. https://github.com/libusb/hidapi/tree/connection-callback

This branch did not modify the code of hid_write. How does it handle the plug-in and removal logic? I'm very sorry because the modification was made 11 months ago. I can't find the equipment to reproduce the problem now, so I can't assist in verifying whether this branch can solve the problem.

Youw, 2025-06-24:
I'm pretty sure you'd hit the same issue with `hid_write` in that branch. I believe @mcuee wants to point out that since you're handling the hot plug/unplug scenarios for your devices, you should have some means to detect the removal/insertion of the device, so instead of making up something of your own, you might as well try that branch.

jianweihao, 2025-06-25:
> I'm pretty sure you'd hit the same issue with `hid_write` in that branch. I believe [@mcuee](https://github.com/mcuee) wants to point out that since you're handling the hot plug/unplug scenarios for [...]

Thank you for your reply. I will do my best to locate the problematic device for testing. However, this process may take a considerable amount of time. Could you please evaluate the rationality of this change from a code perspective? I would be extremely grateful.