libusb/hidapi #756: v0.15.0 crashes when reading from device that went away on Windows 10.

dmaone, 2025-06-28:
Thanks for picking up the maintenance of hidapi. However, I think 0.15.0 is broken now.

Steps to reproduce: 
* open hid device
* periodically call `hid_read()` (I'm using non-blocking version, but I don't see how it'll make things different)
* unplug the device
* observe program crash.

v0.14.0 works in the same scenario.
I tried rewrite `hid_read()` to "remember path, and each time `hid_open_path()` -> `hid_read()` -> `hid_close()`" - I'm missing packets the device sends me, _plus_ the application still crashes - looks like there's no way to detect the situation in advance and avoid crashing.

Youw, 2025-06-28:
Thanks for reporting this.

Can you share a minimal code sample where this is reproducible?

And if you could collect and share a dump file (.dmp) of the crash - that would speed up debugging/investigation _a lot_.

Youw, 2025-06-28:
Please confirm your project respects Notes about multithreading: https://github.com/libusb/hidapi/wiki/Multi%E2%80%90threading-Notes#hid_read